dillonhuff/clockwork #12: Create a lower pass

joyliu37, 2020-04-27:
I was reviewing some of the buffer optimization code in `ubuffer.cpp`. Specifically, functions inside `generate_hls_code` including`generate_prefix`,  `generate_selection`and `generate_bundle`. I think I understand the optimization(bank create, merging, addressing logic creation) happened here. All these passes are happening along with codegen and all the bank/address selection logic are embedded in the `ubuffer` struct after the codegen. 

 I plan to strip the code of ubuffer optimization from the code generation. After this I could get an optimized version of `ubuffer` struct. And then I could use it to create the CGRA config. 

Please let me know if this refactor will affect what you are currently doing. I took a look at your recent commit, it seems you are working on the AWS and FPGA runtime.

dillonhuff, 2020-04-27:
@joyliu37 I think that this can be done without affecting me, but it needs to be done in small steps with frequent merges. We should talk about it more tomorrow.

dillonhuff, 2020-04-27:
@joyliu37 I notice that you have pushed some code to your branch and mention that the tests could run. If so could you create a pull request so I can merge the ubuffer banking method into the master branch?

joyliu37, 2020-04-27:
Yeah, I have some meeting going on this afternoon. I will try to merge the master and create a PR. Thought It's some trivial code change.

joyliu37, 2020-04-27:
Should I keep track of the code generated in soda app?

dillonhuff, 2020-04-27:
No just git ignore it. I will force add the things I want.

dillonhuff, 2020-04-28:
@joyliu37 I like your tiny pull request! Just merged it. I'm going to clean it up a little bit and then we should work on adding support for different inner bank offset computations.

dillonhuff, 2020-04-28:
@joyliu37 Just finished the cleanup. I also added a field to `CodegenOptions` to pick between different bank generation settings.

Could you pull the latest master and let me know that you see the changes?

joyliu37, 2020-04-29:
Yeah. I see the bankOffsetMode, I will continue work on the address generation today.

dillonhuff, 2020-04-29:
@joyliu37 great. LMK if you have any problems. And please push the code you worked on tonight!

dillonhuff, 2020-05-04:
@joyliu37 is the linear bank offset code working now? I notice you have moved on to writing a bank merging pass.

If the new addressing scheme is working could you make a pull request with it?

joyliu37, 2020-05-04:
I could generate the correct string but did not integrated with any test. I feel we should test before merge it.  Do you think it's possible to test this linear address with the HLS backend?

dillonhuff, 2020-05-04:
Yes of course we can test with the HLS backend. Just generate two versions of the HLS code, one with `CodegenOptions` set so that it generates linear addresses, and another with `CodegenOptions` set so that it generates stack addresses. Compile and run both of them and then compare the output streams to ensure that they are the same.

joyliu37, 2020-05-05:
Sounds good, I saw the `run_regression_tb` function automatically run the generate code and check the functionality. Am I correct?
I may need to create a mac unit for the cnn_test. Currently I could just sum all the data together, which is a naive scenarios.

dillonhuff, 2020-05-05:
Yes `run_regression_tb` will generate the test harness, compile it, run it, and then return the sequence of output values from the program as an `std::vector` of strings.

Yes, you should create a mac unit function and put it in `conv_3x3.h`. Multiply and add are both supported by `hw_uint`. LMK if you have problems creating it.

dillonhuff, 2020-05-06:
@joyliu37 were you able to create the unit test? Please make a pull request if so. I see that you have made some significant changes and I don't want us to get out of sync.

joyliu37, 2020-05-06:
Did you mean the unit test for linear address? I did not finish the bank merging stuff for generating the config for memtile. Shall we sync after scrum? I want to know your opinion about the changes that I am currently making.

dillonhuff, 2020-05-06:
Yes, I did mean the unit test for linear addresses. Lets talk after the scrum.

closed 2020-09-21 by dillonhuff