dillonhuff/clockwork #22: CoreIR Codegen for Port Bundle

joyliu37, 2020-06-06:
I found in the current CodeGen you are using a flatten long width bit interface. In order to wire the bundle with correct port, I need to slice or concat. 
I just quick chat with Ross about the port bundle interface. He suggest to use an array of 16bit interface. And that's the formal format for them to do PnR. And it will be easier to `sel` the port inside a wide bundle.
Is these reasonable to you? If so I could do a quick change in the global codegen and make a PR.

dillonhuff, 2020-06-06:
@joyliu37 sure thats fine.

dillonhuff, 2020-06-06:
Also this might be relevant for @jeffsetter since he is generating the compute units for the applications.

jeffsetter, 2020-06-06:
Sounds good to me. However, I believe we have had trouble with CoreIR simulating hierarchical interfaces in the past.

dillonhuff, 2020-06-06:
@joyliu37 can we close this issue now?

closed 2020-06-08 by joyliu37