dillonhuff/clockwork #25: Simple pointwise output values

jeffsetter, 2020-06-11:
I created a simple example that multiplies the input value by 1 and outputs it. The output value I am receiving though is a running sum. The files are attached. Is there some issue with how they are constructed?

[conv_3_3_compute.txt](https://github.com/dillonhuff/clockwork/files/4761860/conv_3_3_compute.txt)
[conv_3_3_memory.txt](https://github.com/dillonhuff/clockwork/files/4761861/conv_3_3_memory.txt)

dillonhuff, 2020-06-11:
@jeffsetter thanks for reporting this.

I don't see any compute unit that multiplies an input by 1 in `conv_3_3_compute.txt`.

Here is what I see in the compute file:

```cpp
#pragma once
#include "hw_classes.h"

//store is: hw_input_copy.stencil(hw_input_copy.s0.x, hw_input_copy.s0.y) = hw_input.stencil(hw_input_copy.s0.x, hw_input_copy.s0.y)
hw_uint<16> hcompute_hw_input_copy_stencil(hw_uint<16>& hw_input_stencil) {
  return hw_input_stencil;
}

//store is: conv.stencil(hw_output.s0.x.xo, hw_output.s0.y.yo) = 0
hw_uint<16> hcompute_conv_stencil() {
  return 0;
}

//store is: conv.stencil(hw_output.s0.x.xo, hw_output.s0.y.yo) = (conv.stencil(hw_output.s0.x.xo, hw_output.s0.y.yo) + int32(hw_input_copy.stencil(hw_output.s0.x.xo, hw_output.s0.y.yo)))
hw_uint<16> hcompute_conv_stencil_1(hw_uint<16>& conv_stencil, hw_uint<16>& hw_input_copy_stencil) {
  return (conv_stencil + int32(hw_input_copy_stencil));
}

//store is: hw_output.stencil(hw_output.s0.x.xo, hw_output.s0.y.yo) = uint8(conv.stencil(hw_output.s0.x.xo, hw_output.s0.y.yo))
hw_uint<16> hcompute_hw_output_stencil(hw_uint<16>& conv_stencil) {
  return uint8(conv_stencil);
}
```

Which compute unit does the multiply by 1? And why is there a compute unit (`hcompute_conv_stencil_1`) that adds one value to another?

jeffsetter, 2020-06-11:
Hm, yea I explained it poorly since Halide simplified the computation. But essentially if you look at the memory and compute files, the output should just be the input file (unless I'm understanding it incorrectly). 

My guess is something odd is happening in hcompute_conv_stencil_1, where the compute should be adding the image_copy to itself. (Where the previous stage sets all initial values to 0).

dillonhuff, 2020-06-11:
Makes sense. Could you send me the halide CPU output for this application?

dillonhuff, 2020-06-11:
Also, when I look at the application the first and second loop nests have different bounds (64 x 64 and 62 x 62 respectively). If this is a pointwise operation why are the bounds different?

jeffsetter, 2020-06-11:
The input image values are: 1 2 3 4 5 6 ...
The cpu output image values are: 1 2 3 4 5 6 ...
The clockwork output image values are: 1 3 6 10 15 21 ...

The Halide app originally was a 3x3 convolution with weights, hence the decreasing loop bound size.
```
        kernel(0,0) = 1;      kernel(1,0) = 0;       kernel(2,0) = 0;
        kernel(0,1) = 0;      kernel(1,1) = 0;       kernel(2,1) = 0;
        kernel(0,2) = 0;      kernel(1,2) = 0;       kernel(2,2) = 0;

        conv(x, y) = 0;

        Func hw_input("hw_input");
        Func hw_input_copy("hw_input_copy");
        hw_input(x, y) = cast<uint16_t>(input(x, y));
        hw_input_copy(x, y) = hw_input(x, y);
        conv(x, y)  += kernel(r.x, r.y) * hw_input_copy(x + r.x, y + r.y);

        Func hw_output("hw_output");
        hw_output(x, y) = cast<uint8_t>(conv(x, y));
        output(x, y) = hw_output(x,y);
```

dillonhuff, 2020-06-11:
@jeffsetter ok I just worked on it a little more, and I think I know what the problem is.

dillonhuff, 2020-06-11:
@jeffsetter I've added that example as a test case and pushed a fix that works for me. Could you pull the latest version and re-try?

jeffsetter, 2020-06-11:
Sweet, it works! (Along with 3x3 convolution with all non-zero weights).

Any insight on what was going wrong?

dillonhuff, 2020-06-11:
Yeah, basically when I was refactoring the algorithm that generates the controller that selects banks to read and write from I introduced an error that wasn't checked by my existing reduce-like test cases. Its hard to explain in text.

closed 2020-06-11 by dillonhuff