dillonhuff/clockwork #30: /misc/fix_barvinok_makefile.py failing

chtsao8, 2020-06-23:
I can't seem to build clockwork using source ./misc/kiwi_setup.sh. Attached is the full output of the build:

[output.txt](https://github.com/dillonhuff/clockwork/files/4816297/output.txt)

Some info: I'm running on Ubuntu 18.04.3. I first tried compiling with gcc-4.8 but switched to gcc-9 for c++17 compatibility. I also believe I should have GMP installed (libgmp10:amd64 is what dpkg tells me), but for some reason it seems barvinok can't find it.

dillonhuff, 2020-06-23:
@chtsao8 thanks for sending this!

Actually from looking at the log neither NTL (the first dependency), nor barvinok can find GMP.

Could you try adding `GMP_PREFIX=/your/path/to/gmp` to the `configure` call for NTL in the script and then rebuilding and sending me the log if it fails.

chtsao8, 2020-06-23:
That did the trick! Thanks for the help.

closed 2020-06-23 by chtsao8