dillonhuff/clockwork #38: Is the sequence of wide fetch width configuration matters?

joyliu37, 2020-07-20:
I am generating the address controller for lake memory tile. 
I am wondering if the sequence of the these controller matters? Should I follow the sequence from smaller starting address to large? 

https://github.com/dillonhuff/clockwork/blob/37a76adc033c9212fd7278e6fd761e77603f4478/lake_controllers/identity_stream/agg2sram.csv#L14-L33

joyliu37, 2020-07-20:
I also did not understand why the `stride_0` is 0 and `stride_1` is not zero. Should we also add `range` to the controller?

mbstrange2, 2020-07-20:
Obviously Dillon would be the best resource for this, but what I do is actually get rid of that loop layer - since it has range 1 and stride 0 it has no actual effect. So we don’t use this loop in our configuration.

kavyasreedhar, 2020-07-20:
+1 to Max's comment. The range in the controller is what Dillon calls "extent". It is shared between the address and valid generators and should be at the top of the file.

dillonhuff, 2020-07-20:
The sequence of controllers does not matter. The reason stride_0 is always
zero and it’s extent is always 1 is because variable 0 is the root variable.

dillonhuff, 2020-07-20:
@dillonhuff sounds good. I'll see you then!

closed 2020-07-21 by dillonhuff