dillonhuff/clockwork #44: Halide Generate Address Dimension Sequence

joyliu37, 2020-07-31:
I want to double check the halide generated clockwork dimension sequence. 
For instance
 `  hcompute_conv_stencil_7->add_load("hw_input_stencil", "7", "(conv_s1_r_x + conv_s1_x)", "(conv_s1_r_y + conv_s1_y)");`
Does this mean `add_load(buf_name, inner_most<-, ->outer most)` 
Or `add_load(buf_name, outer_most<-, ->inner most)` 
I think clockwork follow the C++ array dimension sequence.

jeffsetter, 2020-07-31:
I'm intending that these memory operations have indices from inner-most to outer-most. 
```
add_load(buf_name, inner_most<-, ->outer most)

joyliu37, 2020-07-31:
Could you flip the dimension from outer most to inner most?

jeffsetter, 2020-08-01:
What are you asking? Do you want it to be 
```add_load(buf_name, inner_most<-, ->outer most)```

Or are you asking if you're able to switch the order in Halide?

joyliu37, 2020-08-01:
I want it to be 
`add_load(buf_name, outer_most<-, ->inner most)`

jeffsetter, 2020-08-02:
Changed Halide to generate it in that order.

closed 2020-08-02 by jeffsetter