dillonhuff/clockwork #52: Camera pipeline DSE compute does not match the camera pipeline app

dillonhuff, 2020-08-11:
@jack-melchert @rdaly525 I tried to run camera pipeline using the new dse compute file and I cannot find one of the compute units:

```bash
Checking if hw_input_global_wrapper_stencil_op_hcompute_hw_input_global_wrapper_stencil_0 is an input...
Found compute file for camera_pipeline
ERROR: Could not find Module in namespace!
  Module: hcompute_hw_input_global_wrapper_stencil
  Namespace: global
```

The original camera pipeline had a really weird schedule, which Jeff has now corrected, so perhaps the compute units need to be re-mapped for the more sensible schedule?

jack-melchert, 2020-08-11:
I generated the mapped file from the camera_pipeline in dse_apps, did you try that one?

dillonhuff, 2020-08-11:
@jack-melchert yes and it has the same issue. The problem isn't with the compute units themselves, it is that the compute units in the `dse_apps` folder were generated from a Halide schedule for camera pipeline that doesn't match the one that Jeff is currently using (which can be seen here: https://github.com/dillonhuff/clockwork/blob/68cecd1f5d87df3a0a20d6c615f72de011192cfd/example_progs.cpp#L1631 )

This is related to the issue I brought up in our email thread about the camera pipeline schedule (as of a few days ago) not really making sense.

I would use this file: https://github.com/dillonhuff/clockwork/blob/master/coreir_compute/camera_pipeline_compute.json as the input to the mapper instead.

jack-melchert, 2020-08-11:
Sorry that took so long, I ran into a metamapper bug. I used the new camera_pipeline so hopefully this should work.

dillonhuff, 2020-08-11:
@jack-melchert just tried it. It gets through coreir generation, so thats great!. However I get a bunch of errors about async resets:

```bash
cmd: ${COREIR_PATH}/bin/coreir --inline --load_libs commonlib --input camera_pipeline.json --output camera_pipeline.v
ERROR: {hcompute_corrected_stencil_1_mapped}.i140395651019984_i140395657714128.ASYNCRESET Is not fully connected (N)
{hcompute_corrected_stencil_1_mapped}.i140395651019984_i140395657714128 Is not fully connected (R)
{hcompute_corrected_stencil_1_mapped}.i140395651020112_i140395658446480.ASYNCRESET Is not fully connected (N)
{hcompute_corrected_stencil_1_mapped}.i140395651020112_i140395658446480 Is not fully connected (R)
{hcompute_corrected_stencil_1_mapped}.i140395651020304_i140395652650576.ASYNCRESET Is not fully connected (N)
{hcompute_corrected_stencil_1_mapped}.i140395651020304_i140395652650576 Is not fully connected (R)
{hcompute_corrected_stencil_1_mapped}.i140395651020432_i140395652923536.ASYNCRESET Is not fully connected (N)
{hcompute_corrected_stencil_1_mapped}.i140395651020432_i140395652923536 Is not fully connected (R)
{hcompute_corrected_stencil_1_mapped}.i140395651020560_i140395653210640.ASYNCRESET Is not fully connected (N)
{hcompute_corrected_stencil_1_mapped}.i140395651020560_i140395653210640 Is not fully connected (R)
{hcompute_corrected_stencil_1_mapped}.i140395651020624_i140395652923536.ASYNCRESET Is not fully connected (N)
{hcompute_corrected_stencil_1_mapped}.i140395651020624_i140395652923536 Is not fully connected (R)
{hcompute_corrected_stencil_1_mapped}.i140395651021264_i140395657714128.ASYNCRESET Is not fully connected (N)
{hcompute_corrected_stencil_1_mapped}.i140395651021264_i140395657714128 Is not fully connected (R)
{hcompute_corrected_stencil_1_mapped}.i140395651021840_i140395652650576.ASYNCRESET Is not fully connected (N)
{hcompute_corrected_stencil_1_mapped}.i140395651021840_i140395652650576 Is not fully connected (R)
{hcompute_corrected_stencil_1_mapped}.i140395651022416_i140395657714128.ASYNCRESET Is not fully connected (N)
{hcompute_corrected_stencil_1_mapped}.i140395651022416_i140395657714128 Is not fully connected (R)
{hcompute_corrected_stencil_1_mapped}.i140395651096656_i140395652923536.ASYNCRESET Is not fully connected (N)
{hcompute_corrected_stencil_1_mapped}.i140395651096656_i140395652923536 Is not fully connected (R)
{hcompute_corrected_stencil_1_mapped}.i140395651097360_i140395652923536.ASYNCRESET Is not fully connected (N)
{hcompute_corrected_stencil_1_mapped}.i140395651097360_i140395652923536 Is not fully connected (R)
{hcompute_corrected_stencil_1_mapped}.i140395651098576_i140395652923536.ASYNCRESET Is not fully connected (N)
{hcompute_corrected_stencil_1_mapped}.i140395651098576_i140395652923536 Is not fully connected (R)
```

I don't think the CGRA is actually going to support async reset for individual registers on the fabric, so I'm not sure what the fix is.

jack-melchert, 2020-08-11:
Hm yeah that ASYNCRESET signal seems like an auto-generated signal for every magma (maybe every peak?) circuit. Its not really being used for anything so maybe there is a way to disable it and remove it from the PE. I'll look into it.

dillonhuff, 2020-08-11:
@jack-melchert sounds good. LMK when you push a new version.

jack-melchert, 2020-08-11:
Just pushed the updated version, hopefully it works now.

dillonhuff, 2020-08-11:
@jack-melchert @rdaly525 my testbench for it passed! I'm actually a bit nervous that that just worked so easily.

Did the mapping change the latencies of any of the compute units? I'm assuming all latencies are zero except the roms, which have latency 1.

jack-melchert, 2020-08-11:
Yep, all compute units have a latency of 0 for now. Thats great news!

dillonhuff, 2020-08-11:
Well thats good. Have you added latency info to the compute unit file so that I can pull it out automatically?

jack-melchert, 2020-08-11:
Nope not yet, I'll check with ross to see what the status of that part is.

rdaly525, 2020-08-12:
There is a branch on the metamapper that accomplishes this, but minor bugs generating coreir

jack-melchert, 2020-08-12:
Dillon what exactly does your testbench cover? Does it pass it through place and route or is it before that stage?

dillonhuff, 2020-08-12:
@jack-melchert its a test right before place and route. The next step is to actually place and route it. How come?

jack-melchert, 2020-08-12:
Great, just wanted to know so I can keep Priyanka and Kathleen up to date on our progress

dillonhuff, 2020-08-12:
Cool, should I try to place, route, and run it now? Or are there other changes you'd like to make first?

Also is there a comprehensive list of apps you want to run for the paper?

jack-melchert, 2020-08-12:
Yes it would be awesome if you could try place and routing! Currently we are just trying to get the other apps through Ross's metamapper. The apps we are targeting right now are all in the dse_apps folder. When I get the other ones through halide to hardware I will add them to that folder

dillonhuff, 2020-08-12:
@jack-melchert what branches of garnet (and other repos that generate the design) are you using? I assume anything with custom PEs will need custom verilog?

jack-melchert, 2020-08-12:
Ok so we are using this branch of garnet https://github.com/StanfordAHA/garnet/tree/dse-cgra-without-lassen
It requires this repo https://github.com/jack-melchert/peak_generator
Unfortunately the peak_generator repo has requirement for a custom version of ast_tools, so it might be easiest to create a new python virtual environment before installing the requirements.
Let me know if you run into any issues and I will help fix them

dillonhuff, 2020-08-12:
@jack-melchert so would it be best to build this using the aha flow docker container, but just switching to the `dse-cgra-without-lassen` branch of garnet?

Just as background: has anyone placed, routed, and run anything using this DSE flow before?

jack-melchert, 2020-08-12:
Sure it might be easier using docker. Here is a new script to get it all installed 
```
# install the aha wrapper script
pip install -e .

# pull docker image from docker hub (this takes a while!)
docker pull stanfordaha/garnet:latest

# run the container in the background and delete it when it exits
# (this will print out the name of the container to attach to)
aha docker

# you can also do the following instead, but using the wrapper shim is
# suggested as it will do things like mount /cad automatically when 
# detected
# docker run -d -it --rm stanfordaha/garnet bash

# attach to the container name printed out above
docker attach <container-name>

# load required simulator (ncsim)
module load incisive

rm -rf ast_tools/
git clone https://github.com/jack-melchert/ast_tools.git
cd ast_tools/
pip install -e .
cd ../

git clone https://github.com/jack-melchert/peak_generator.git
cd peak_generator/
pip install -e .
cd ../

cd garnet/
git fetch
git checkout dse-cgra-without-lassen
cd ../

# run a small regression suite
aha regress fast
```

It needs the info from the mapper to go through pnr successfully, so this script will error our on the `aha regress fast` command

dillonhuff, 2020-08-12:
@jack-melchert thanks! Just got it installed, and got to the failure you mentioned.

A few things:
* Could you generate mapped compute files for pointwise? It'll be simpler for me to get the flow working with a smaller example (less time to generate the fabric, and its easy to eye check the output)
* What is the fix for this mapper error? What other info does pnr need that it doesn't currently get?

jack-melchert, 2020-08-12:
Sure I will work on creating that mapped compute file now. 

I am also looking into this mapper error. I think its simply getting the mapper to correctly generate the netlist and bus for custom PEs. I'm not sure how hard coded the information coming from Keyi's mapper is.

jack-melchert, 2020-08-12:
So I've updated garnet and clockwork with the new PE for the pointwise app. You should just need to pull both of those repos.

dillonhuff, 2020-08-12:
@jack-melchert I'm working on it now.

dillonhuff, 2020-08-12:
@jack-melchert apparently the "arst" issue is back. When I try to map:

```bash
cmd: ${COREIR_PATH}/bin/coreir --inline --load_libs commonlib --input pointwise.json --output pointwise.v
ERROR: {cu_op_hcompute_mult_stencil}.inner_compute.arst Is not fully connected (N)
{cu_op_hcompute_mult_stencil}.inner_compute Is not fully connected (R)

ERROR: {cu_op_hcompute_mult_stencil}.inner_compute.arst Is not fully connected (N)
{cu_op_hcompute_mult_stencil}.inner_compute Is not fully connected (R)
```

jack-melchert, 2020-08-12:
Ah thats annoying. I'll check with Ross and see if he knows why that has come back.

rdaly525, 2020-08-12:
@dillonhuff can you change the command to ${COREIR_PATH}/bin/coreir --inline --load_libs commonlib --input pointwise.json --output pointwise.v -p "wireclocks-arst; wireclocks-clk"

dillonhuff, 2020-08-12:
@rdaly525 I did that and it fixed the verilog generation for pointwise. Thanks!

However, there is no arst on the fabric for each tile right? Does that port actually need to be connected when we place and route on the fabric?

rdaly525, 2020-08-12:
That is correct, there is no routable arst. Although I think there is a literal arst for each tile in the verilog design of the CGRA.

dillonhuff, 2020-08-12:
@rdaly525 so should we delete the arst fields on each tile during mapping?

@jack-melchert where in the PE instance is the instruction configuration? Is it the value of this constant connected to this: https://github.com/dillonhuff/clockwork/blob/2522735892fe45a1bd36465849d5f0fb2a3b8977/coreir_apps/dse_paper/pointwise.json#L1891

rdaly525, 2020-08-12:
Ah I think there was some miscommunication. @jack-melchert, you should not be calling the wireclocks passes unless you are generating verilog.

jack-melchert, 2020-08-12:
Whats the alternative? Garnet generates verilog so this issue will show up unless we have a workaround.

@dillonhuff Yep thats correct, the "inst" signal is the instruction config.

rdaly525, 2020-08-12:
I am just saying, after mapping an application do not call the wireclocks passes (unless we are testing the mapped application via verilog). The PEs themselves in the mapped_app will have clk/arst ports and that is unavoidable.

dillonhuff, 2020-08-12:
@rdaly525 @jack-melchert well I can at least get the thing to place, though routing is failing. BTW do names of ports on `WrappedPE_wrapped` correspond exactly to names of ports on the PE tile?

jack-melchert, 2020-08-12:
Yes I believe that's correct.

dillonhuff, 2020-08-12:
@jack-melchert so the PE has an `inst` port and we should place the instruction constant on the fabric and route it to the `inst` port?

jack-melchert, 2020-08-12:
Actually there is a special name for the instr port when referring to the full PE tile, its called "config". Sorry about that, didn't realize until I wrote a test.

jack-melchert, 2020-08-13:
Hey Dillon, just curious if you got the place and route in garnet working?

jack-melchert, 2020-08-14:
Also, is it possible for me to run your pre-pnr verilog testbench? We're thinking of pushing that through our power flow to get some preliminary evaluation numbers.

dillonhuff, 2020-08-14:
@jack-melchert I have not PnRed it yet, thats going to take more work.

You could definitely run the pre-pnr testbench. I've added versions with raw address generators + dual port srams here: https://github.com/dillonhuff/clockwork/tree/master/coreir_apps/dse_raw_sram

jack-melchert, 2020-08-14:
Great, thanks! For now we are planning on going with pre-PnR evaluation, so PnR shouldn't be a bottleneck for the DSE paper at least.

dillonhuff, 2020-08-14:
@jack-melchert thats good to hear! I'm working with Max to get a tile with built in addrgen too, so LMK if you want a different tile.

jack-melchert, 2020-08-14:
Would it be possible for you to teach me how to generate the testbench as well? I really only care about the mapped verilog with PE and memory tiles, but the testbench would be nice for verification purposes.

dillonhuff, 2020-08-14:
@jack-melchert sure, if you have clockwork downloaded you can just do:

```
./rebuild_and_run.sh cgra-flow
```

and it will generate and run the testbench. You could also make a custom command and add it so that you have your own flow. LMK if you want to chat about it via zoom.

jack-melchert, 2020-08-14:
That seems to be working for me! For camera pipeline, I get some errors about unconnected wires in the rom. Do you see the same errors?

jack-melchert, 2020-08-14:
Maybe it would be a good idea to have a chat over zoom to go over how I would set this up for a new app. Let me know if you are free today anytime before 5pm (outside of the scrum).

dillonhuff, 2020-08-14:
@jack-melchert I'm free right after the scrum, so we could meet then

jack-melchert, 2020-08-15:
Awesome its working now! Thanks for your help, also you probably want to remove all of those assert(false) things, they are still in master

dillonhuff, 2020-08-15:
@jack-melchert nice!

LMK if you are going to need the DNNs, since that will take more work on my end.

jack-melchert, 2020-08-16:
So Kathleen is running into an issue with the coreir_mem verilog modules have a init parameter in the wrong place, not sure if this is an issue that can be fixed in this flow though. Do you know if this might be fixed with a change to a memory tile that has actual sram macros?

dillonhuff, 2020-08-16:
@jack-melchert its probably related to her coreir version. Maybe it is this issue: https://github.com/rdaly525/coreir/issues/938 https://github.com/rdaly525/coreir/issues/932

dillonhuff, 2020-08-16:
@jack-melchert is the verilog malformed? Or is it a coreir error?

jack-melchert, 2020-08-16:
The verilog is malformed, the coreir to verilog works just fine. I'll update my coreir version and generate the verilog again

dillonhuff, 2020-08-16:
@jack-melchert on which application? Could you send me the text of the error?

jack-melchert, 2020-08-16:
Its on camera pipeline, in the verilog I generate with ./rebuld_and_run cgra_flow. 
```
 3390 module coreir_mem #(
 3391     parameter has_init = 1'b0,
 3392     parameter sync_read = 1'b0,
 3393     parameter depth = 1,
 3394     parameter width = 1
 3395 ) (
 3396     input clk,
 3397     input [width-1:0] wdata,
 3398     input [$clog2(depth)-1:0] waddr,
 3399     input wen,
 3400     output [width-1:0] rdata,
 3401     input [$clog2(depth)-1:0] raddr
 3402 );
 3403   reg [width-1:0] data [depth-1:0];
 3404   parameter [width*depth-1:0] init = 0;
```
On line 3404 of the verilog, that init should be moved into the parameter block at the top

dillonhuff, 2020-08-16:
@jack-melchert that looks like the actual coreir primitive has a verilog generation error. Have you asked @rdaly525 about this?

jack-melchert, 2020-08-16:
That makes sense, I'll follow up with him tomorrow morning

closed 2020-09-21 by dillonhuff