dillonhuff/clockwork #75: Make a smaller version of mobilenet?

dillonhuff, 2020-09-28:
@jeffsetter I can schedule the current `mobilenet`, but I cannot actually implement it with the 16 bit controllers we have because the total number of loop iterations for some of the loops is over 100,000.

The whole application takes around 250k cycles with the current scheduler. Could you cut down the tile sizes (maybe by a factor of 3 in each direction) so that we can test it as a smaller application?

Thanks!

jeffsetter, 2020-09-28:
I'll create an example that is smaller, but we might want to increase the loop iterators as larger than 16bits, since having 100k cycles isn't that uncommon for these tiled applications.

dillonhuff, 2020-09-28:
@jeffsetter thanks man. I totally agree that we need wider control registers.

closed 2020-09-29 by dillonhuff