dillonhuff/clockwork #83: Can I remove this extra one cycle delay for single port garnet test

joyliu37, 2020-10-15:
https://github.com/dillonhuff/clockwork/blob/1d3ced992f8fc86261eee46e35fa498c6adb363d/build_set_test.cpp#L16041-L16048

joyliu37, 2020-10-15:
Is this just conservative or it has some special purpose?

dillonhuff, 2020-10-15:
Just leave it in. It is adding 1 cycle to the latency of the *entire application* so it really does not matter for performance.

joyliu37, 2020-10-15:
So is this one cycle latency capture by any shift-register? Since we need to add that delay between input and the first buffer.

dillonhuff, 2020-10-15:
No, the compiler assumes that the controller outside the fabric that is sending data to the fabric is aware of the schedule, so there is no need for a shift register. If the schedule says the first data will be read from the input on cycle 20, then the controller for the fabric should just send in the first data on cycle 20.

Did that answer your question?

joyliu37, 2020-10-16:
> No, the compiler assumes that the controller outside the fabric that is sending data to the fabric is aware of the schedule, 

I think this is not true for current garnet backend. Because we did not send any schedule information to the test bench which is generated by halide. 
I think I could fix this by substracting the lexmin of the global 1D schedule when generating lake configuration.

closed 2020-10-17 by dillonhuff