dillonhuff/clockwork #92: Coreir error in harris_sch1

dillonhuff, 2020-11-03:
@jeffsetter apparently something is malformed in the coreir for harris_sch1:

```
ERROR: const_n255__1148$11 already an instance

/home/dhuff/clockwork/lib/libcoreir.so(_ZN6CoreIR9ModuleDef11addInstanceENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEPNS_6ModuleESt3mapIS6_PNS_5ValueESt4lessIS6_ESaISt4pairIKS6_SB_EEE+0x7a)[0x7ff26641b43c]
/home/dhuff/clockwork/lib/libclkwrk.so(_Z8ConstDupPN6CoreIR8InstanceE+0x280)[0x7ff2679f65e4]
/home/dhuff/clockwork/lib/libcoreir.so(_ZN6CoreIR19InstanceVisitorPass17runOnGenInstancesEPNS_9GeneratorERSt3setIPNS_8InstanceESt4lessIS5_ESaIS5_EE+0xd5)[0x7ff26644fcd5]
/home/dhuff/clockwork/lib/libcoreir.so(_ZN6CoreIR11PassManager22runInstanceVisitorPassEPNS_4PassE+0x1fd)[0x7ff2664554c3]
/home/dhuff/clockwork/lib/libcoreir.so(_ZN6CoreIR11PassManager7runPassEPNS_4PassERSt6vectorINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESaIS9_EE+0x378)[0x7ff266455b02]
/home/dhuff/clockwork/lib/libcoreir.so(_ZN6CoreIR11PassManager3runERSt6vectorINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESaIS7_EES9_+0x5b0)[0x7ff266456ac8]
/home/dhuff/clockwork/lib/libcoreir.so(_ZN6CoreIR7Context9runPassesESt6vectorINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESaIS7_EES9_+0x7e)[0x7ff2663963c0]
/home/dhuff/clockwork/lib/libclkwrk.so(_Z17garnet_map_modulePN6CoreIR6ModuleE+0xb2c)[0x7ff2679f7a90]
/home/dhuff/clockwork/lib/libclkwrk.so(_Z15generate_coreirR14CodegenOptionsRSt3mapINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE7UBufferSt4lessIS7_ESaISt4pairIKS7_S8_EEER4progP13isl_union_mapR13schedule_info+0x2aa)[0x7ff2679f91ed]
./clockwork(+0x214742)[0x5562b2c0e742]
./clockwork(+0x214818)[0x5562b2c0e818]
./clockwork(+0x217d83)[0x5562b2c11d83]
./clockwork(+0x21ace2)[0x5562b2c14ce2]
./clockwork(+0x237287)[0x5562b2c31287]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xe7)[0x7ff264eafb97]
./clockwork(+0x11c46a)[0x5562b2b1646a]
dhuff@kiwi:~/clockwork$ 
```

The instance in question seems to be in the compute file:

```bash
dhuff@kiwi:~/clockwork$ grep -r "const_n255__1148\$11" ./coreir_compute/harris_sch1_compute.json 
          "const_n255__1148$11":{
          ["smax_1274_1148_1275.in1","const_n255__1148$11.out"],
```

Any idea whats up here? Several of the other harris variants run without issue.

jeffsetter, 2020-11-03:
Looking at the compute file, `const_n255__1148$11` is correctly created just one time. 

It looks like you might be running a constant duplication pass?
My guess is that the duplication pass is trying to create another constant of a similar name, and it ended up choosing a name that already exists?

dillonhuff, 2020-11-03:
@jeffsetter something is very odd about this harris variant. As you suggested turning off constant duplication in mapping removes this error, but then I get a new error:

```
ERROR: size: 3

/home/dhuff/clockwork/lib/libclkwrk.so(_Z12ConstReplacePN6CoreIR8InstanceE+0x115)[0x7f3b667b8e82]
/home/dhuff/clockwork/lib/libcoreir.so(_ZN6CoreIR19InstanceVisitorPass17runOnModInstancesEPNS_6ModuleERSt3setIPNS_8InstanceESt4lessIS5_ESaIS5_EE+0xcf)[0x7f3b65213bd5]
/home/dhuff/clockwork/lib/libcoreir.so(_ZN6CoreIR11PassManager22runInstanceVisitorPassEPNS_4PassE+0x13c)[0x7f3b65219402]
/home/dhuff/clockwork/lib/libcoreir.so(_ZN6CoreIR11PassManager7runPassEPNS_4PassERSt6vectorINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESaIS9_EE+0x378)[0x7f3b65219b02]
/home/dhuff/clockwork/lib/libcoreir.so(_ZN6CoreIR11PassManager3runERSt6vectorINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESaIS7_EES9_+0x5b0)[0x7f3b6521aac8]
/home/dhuff/clockwork/lib/libcoreir.so(_ZN6CoreIR7Context9runPassesESt6vectorINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESaIS7_EES9_+0x7e)[0x7f3b6515a3c0]
/home/dhuff/clockwork/lib/libclkwrk.so(_Z17garnet_map_modulePN6CoreIR6ModuleE+0xb53)[0x7f3b667bba07]
/home/dhuff/clockwork/lib/libclkwrk.so(_Z15generate_coreirR14CodegenOptionsRSt3mapINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE7UBufferSt4lessIS7_ESaISt4pairIKS7_S8_EEER4progP13isl_union_mapR13schedule_info+0x2aa)[0x7f3b667bce93]
./clockwork(+0x215574)[0x558146e6f574]
./clockwork(+0x21564a)[0x558146e6f64a]
./clockwork(+0x218b76)[0x558146e72b76]
./clockwork(+0x21bad5)[0x558146e75ad5]
./clockwork(+0x2380b1)[0x558146e920b1]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xe7)[0x7f3b63c73b97]
./clockwork(+0x11c4ca)[0x558146d764ca]
```

And when I comment out the mapping pass altogether the design compiles to verilog and simulates, but the output does not match the CPU result. None of the other Harris variants have any of these problems.

Any idea why this particular variant might have so many issues?

jeffsetter, 2020-11-03:
I'm not sure why it's causing so many issues, but this schedule basically creates no intermediate buffers. So the computation kernel is rather large.

The header compute file seems to work, so there must be an issue with the coirer compute.

dillonhuff, 2020-11-06:
@jeffsetter I've cut out the load loop and just reduced the entire application to the one statement that uses the compute unit to isolate the problem. From what I can see the coreir compute unit always produces 0, while the CPU compute unit does not.

Could you take a look at the coreir compute generation you are doing to see if this is the case? Any help would be appreciated.

jeffsetter, 2020-11-08:
I took a look at the compute. I was able to create a visual representation, meaning all of the connections make sense, but it hard to identify if there are any issues. 

Btw, the compute is only intended to create a few 255 (representing corners) and all other output values are 0. I might be easier to debug if you end the computation right before the non-maximal suppression (all the compares).

dillonhuff, 2020-11-09:
I have alrea

dillonhuff, 2020-11-09:
@jeffsetter there are several cases where the Coreir compute unit produces a zero output but the CPU compute unit produces 255. Could you add a testbench to your code where you compare the two different compute units values and let me know if you find any differences? It only takes a few dozen random inputs on my end to get a mismatch between the CPU and coreir compute units.

dillonhuff, 2020-11-11:
@jeffsetter I have written a compute unit regression test generator and used it to test all of the harris variants compute units.

closed 2020-11-11 by dillonhuff