dillonhuff/clockwork #93: harris_sch6 has a buffer that is not an input but has no writers?

dillonhuff, 2020-11-03:
@jeffsetter when I run 'prg.sanity_check()` on `harris_sch6` I get:

```bash
Error: grad_x_unclamp_stencil has 2 readers, but it is never written.
```

Should this buffer be an input?

jeffsetter, 2020-11-03:
There was a weird error in my Halide schedule where `grad_x_unclamp` wasn't being produced. I was able to fix it in https://github.com/dillonhuff/clockwork/pull/94

dillonhuff, 2020-11-03:
@jeffsetter thanks! I just merged. Now I get the error:

```bash
op_hcompute_lgxy_stencil is not a producer or consumer of any dependency
```

jeffsetter, 2020-11-03:
What does that mean? I see that it is used here:
```//store is: lgxy.stencil(((lgxy_s0_x_x*2) + -1), lgxy_s0_y) = (int16)0
  auto hcompute_lgxy_stencil = lgxy_s0_x_x->add_op("op_hcompute_lgxy_stencil");
  hcompute_lgxy_stencil->add_function("hcompute_lgxy_stencil");
  prg.buffer_port_widths["lgxy_stencil"] = 16;
  hcompute_lgxy_stencil->add_store("lgxy_stencil", "lgxy_s0_y", "((lgxy_s0_x_x*2) + -1)");
```

And later hcompute_lgxy_stencil_2 uses it 
`hcompute_lgxy_stencil_2->add_load("lgxy_stencil", "lgxy_s1_y", "lgxy_s1_x");`

Is there an issue with this store in between?
```//store is: lgxy.stencil((lgxy_s0_x_x*2), lgxy_s0_y) = (int16)0
  auto hcompute_lgxy_stencil_1 = lgxy_s0_x_x->add_op("op_hcompute_lgxy_stencil_1");
  hcompute_lgxy_stencil_1->add_function("hcompute_lgxy_stencil_1");
  hcompute_lgxy_stencil_1->add_store("lgxy_stencil", "lgxy_s0_y", "(lgxy_s0_x_x*2)");
  auto lgxy_s1_y = prg.add_loop("lgxy_s1_y", -1, 59);
  auto lgxy_s1_x = lgxy_s1_y->add_loop("lgxy_s1_x", -1, 59);
```

closed 2020-11-10 by dillonhuff