dillonhuff/clockwork #100: Malfored ROM init in one of the harris compute units?

dillonhuff, 2020-11-10:
@jeffsetter I'm getting an error:

```bash
cmd: ${COREIR_PATH}/bin/coreir --inline --load_libs commonlib,cgralib --input harris_sch4_1pp3c.json --output harris_sch4_1pp3c.v -p "rungenerators; wireclocks-arst; wireclocks-clk"
ERROR: Got non-number for json array element in mem init
```

In the harris_sch4_1ppc3c compute unit file. There seems to be a ROM in the compute unit file that has `null` values in its init.

```bash
dhuff@kiwi:~/clockwork$ grep "init" ./coreir_compute/harris_sch4_1pp3c_compute.json 
            "modargs":{"init":["Json",[-1,null,1,-2,null,2,-1,null,1]]}
            "modargs":{"init":["Json",[-1,null,1,-2,null,2,-1,null,1]]}
            "modargs":{"init":["Json",[-1,null,1,-2,null,2,-1,null,1]]}
            "modargs":{"init":["Json",[1,2,1,null,null,null,-1,-2,-1]]}
            "modargs":{"init":["Json",[1,2,1,null,null,null,-1,-2,-1]]}
            "modargs":{"init":["Json",[1,2,1,null,null,null,-1,-2,-1]]}
```

dillonhuff, 2020-11-10:
@jeffsetter there is a similar issue in another of the implementations:

```bash
dhuff@kiwi:~/clockwork$ grep "init" ./coreir_compute/harris_sch3_1pp9c_compute.json 
            "modargs":{"init":["Json",[-1,null,1,-2,null,2,-1,null,1]]}
            "modargs":{"init":["Json",[1,2,1,null,null,null,-1,-2,-1]]}
```

jeffsetter, 2020-11-10:
Good catch. The original init values were not collected, so I'm not making sure all init values are set. Fixed in https://github.com/dillonhuff/clockwork/pull/102.

closed 2020-11-10 by dillonhuff