dillonhuff/clockwork #112: In prog: harris_sch4_1pp3c compute unit: hcompute_cim_stencil has a mismatch between C++ and coreir

dillonhuff, 2020-11-13:
@jeffsetter @rdaly525 I'm guessing this is related to #108 but there is an error in one of the Harris sch4 compute units:

```bash
ERROR in compute unit: hcompute_cim_stencil
in0_lgxx_stencil[0] -> 18547
in0_lgxx_stencil[0] -> 0100100001110011
in1_lgxy_stencil[0] -> 56401
in1_lgxy_stencil[0] -> 1101110001010001
in2_lgyy_stencil[0] -> 23807
in2_lgyy_stencil[0] -> 0101110011111111
	coreir_result: 18585
	cpp_result   : 22681
```

Could you take a look at this?

Testbench is here: https://github.com/dillonhuff/clockwork/tree/master/compute_unit_regressions/harris_sch4_1pp3c/hcompute_cim_stencil

jeffsetter, 2020-11-13:
Can confirm. After modifying "wire" to "wire signed", it passes.

rdaly525, 2020-11-17:
Sorry, just saw this. Ill take a look.