dillonhuff/clockwork #130: Question about shift register optimization pass in coreIR backend

joyliu37, 2020-12-01:
I am refactoring my shift-register optimization pass. (`generate_bank_and_merge`, `port_group_banks`)
Eventually the difference between single port mem and dual port is just the bank grouping.
I am going over some of the code in https://github.com/dillonhuff/clockwork/blob/4fc749af30ec300ac15eeb27d476eaa313351b19/coreir_backend.cpp#L4501.
I am wondering if this method is stable? I could leverage some of the logic inside. Why we have two separate pass inside this function?

dillonhuff, 2020-12-01:
@joyliu37 its not that stable, but the data structure `dgraph` is. You could just copy the logic you want for now.

joyliu37, 2020-12-01:
Sounds good. I will use `dgraph` and `ubuffer_impl`. I think it's a good idea to have some separate data structure carrying the hardware graph structure, instead of saving that inside `ubuffer.bank_list`

closed 2020-12-11 by joyliu37