dillonhuff/clockwork #140: Issues with dynamic stores/loads

jeffsetter, 2021-01-25:
I tried making a histogram test, but am having issues with the unoptimized version running. The error I encounter is:

```
bin/unoptimized_histogram.cpp: In function ‘void op_hcompute_histogram_stencil_1(bin_stencil_cache&, histogram_stencil_cache&, int, int, int, int)’:
bin/unoptimized_histogram.cpp:450:114: error: invalid initialization of reference of type ‘hw_uint<16>&’ from expression of type ‘hw_uint<64>’
  auto compute_result = hcompute_histogram_stencil_1(bin_stencil_histogram_s1_r_x_value, histogram_stencil_0_value);
                                                                                                                  ^
In file included from bin/unoptimized_histogram.cpp:10:0:
bin/histogram_compute.h:28:13: note: in passing argument 1 of ‘hw_uint<16> hcompute_histogram_stencil_1(hw_uint<16>&)’
 hw_uint<16> hcompute_histogram_stencil_1(hw_uint<16>& histogram_stencil) {
             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
At global scope:
cc1plus: warning: unrecognized command line option ‘-Wno-unknown-warning-option’
../../hw_support/hardware_targets.mk:151: recipe for target 'bin/unoptimized_histogram.o' failed
make[1]: *** [bin/unoptimized_histogram.o] Error 1
make[1]: Leaving directory '/nobackup/setter/h2h/hwbuffer/polyir/apps/hardware_benchmarks/tests/histogram'
../../hw_support/hardware_targets.mk:331: recipe for target 'bin/output_clockwork.png' failed
make: *** [bin/output_clockwork.png] Error 2
```
Attached are the files that I used to create the histogram test.
[histogram_compute.txt](https://github.com/dillonhuff/clockwork/files/5868624/histogram_compute.txt)
[histogram_memory.txt](https://github.com/dillonhuff/clockwork/files/5868625/histogram_memory.txt)

dillonhuff, 2021-02-01:
@jeffsetter could you re-name this application? We already have an app called `histogram` and a corresponding compute file, so I'd have to overwrite that one in the repo to test this one.

jeffsetter, 2021-02-01:
Do you mean changing the line `prg.name = "histogram";` to something else?

dillonhuff, 2021-02-01:
yes, and changing whatever other names you need to change in order for the compute unit file to have a different name.

jeffsetter, 2021-02-01:
[histogram1d_compute.txt](https://github.com/dillonhuff/clockwork/files/5906261/histogram1d_compute.txt)
[histogram1d_memory.txt](https://github.com/dillonhuff/clockwork/files/5906263/histogram1d_memory.txt)

dillonhuff, 2021-02-05:
@jeffsetter thanks!

Having looked into this more I'd recommend a temporary solution: When you generate the compute units that receive data from a dynamic load just add dummy arguments to the compute units (one for each buffer that is used in the dynamic address calculation). That should silence this bug.

I only ever implemented one example that uses dynamic access as a demo (see here: https://github.com/dillonhuff/clockwork/blob/87fae4cd3cef5e521425cb4a550d114c026f8788/build_set_test.cpp#L11922) and I had to use this trick. Unfortunately I don't see a convenient way to get out of using this hack for the time being.

jeffsetter, 2021-03-02:
The above worked for simple dynamic stores/loads.

However, the more recent example (bilateral_grid) has multiple dimensions that are dynamically loaded/stored. Is there a suggestion on how create this? Below is the error I'm seeing.
```
g++-7 -std=c++17 -I../../../../../clockwork -I../../../../../clockwork/include -I../../../../../clockwork/barvinok-0.41/isl -fPIC -c bin/clockwork_codegen.cpp -o bin/clockwork_codegen.o
In file included from bin/clockwork_codegen.cpp:2:0:
bin/bilateral_memory.cpp: In function ‘prog bilateral()’:
bin/bilateral_memory.cpp:234:201: error: no matching function for call to ‘ir_node::add_dynamic_load(const char [14], const char [15], const char [18], const char [18], const char [15], const char [18], $
 y", "interpolated_s0_x", "yii_p1_stencil", "interpolated_s0_y", "xii_stencil", "interpolated_s0_x");
                                                                                                   ^
In file included from bin/bilateral_memory.cpp:4:0,
                 from bin/clockwork_codegen.cpp:2:
../../../../../clockwork/prog.h:583:8: note: candidate: void ir_node::add_dynamic_load(const string&, const string&, const string&)
   void add_dynamic_load(const std::string& buf,
        ^~~~~~~~~~~~~~~~
../../../../../clockwork/prog.h:583:8: note:   candidate expects 3 arguments, 8 provided
```

The associated files are below:
[bilateral_compute.txt](https://github.com/dillonhuff/clockwork/files/6065186/bilateral_compute.txt)
[bilateral_memory.txt](https://github.com/dillonhuff/clockwork/files/6065187/bilateral_memory.txt)

dillonhuff, 2021-03-03:
@jeffsetter you could try flattening the access pattern to a 1D address. clockwork has no support for multi-dimensional dynamic accesses.