dillonhuff/clockwork #152: FFT8 doesn't pass memory mapper

kosho2013, 2021-03-06:
@joyliu37 
FFT8 doesn't pass memory mapper, some issues with buffer.

**clockwork: ubuffer.cpp:4343: bank UBuffer::compute_bank_info(uset*, isl_point*, std::set<std::__cxx11::basic_string<char> >, std::set<std::__cxx11::basic_string<char> >): Assertion `false' failed.
./rebuild_and_run.sh: line 8: 21439 Aborted                 (core dumped) ./clockwork $1**

joyliu37, 2021-03-09:
This is supposed to be a banking issue. 
Which branch are you in and what's the command to reproduce this issue?

kosho2013, 2021-03-09:
My branch is called "kosho".
The command is "./rebuild_and_run.sh lake-exp".
You can modify the apps in "build_set_test.cpp" to experiment with different unrolling schemes.