dillonhuff/clockwork #162: Boundary Cases

jeffsetter, 2021-03-31:
I was wondering if these outputs from Halide were valid in clockwork. And if they properly map to the FPGA (I already know that the CGRA has issues with these piecewise stores).

I specifically have concerns about:
```hcompute_repeat_image_stencil->add_load("hw_input_global_wrapper_stencil", "63", "select(((64 <= repeat_image_s0_x) || (repeat_image_s0_x < 0)), (repeat_image_s0_x % 64), max(min(repeat_image_s0_x, 63), 0))");```

and
```hcompute_mirror_image_stencil->add_load("hw_input_global_wrapper_stencil", "0", "select(((64 <= mirror_image_s0_x) || (mirror_image_s0_x < 0)), min(select(((mirror_image_s0_x % 128) < 64),(mirror_image_s0_x % 128), (127 - (mirror_image_s0_x % 128))), 63), max(min(mirror_image_s0_x, 63), 0))");```

Here are the full applications:
[bc_const_compute.txt](https://github.com/dillonhuff/clockwork/files/6238159/bc_const_compute.txt)
[bc_const_memory.txt](https://github.com/dillonhuff/clockwork/files/6238160/bc_const_memory.txt)
[bc_repeat_compute.txt](https://github.com/dillonhuff/clockwork/files/6238164/bc_repeat_compute.txt)
[bc_repeat_memory.txt](https://github.com/dillonhuff/clockwork/files/6238165/bc_repeat_memory.txt)[bc_mirror_compute.txt](https://github.com/dillonhuff/clockwork/files/6238162/bc_mirror_compute.txt)
[bc_mirror_memory.txt](https://github.com/dillonhuff/clockwork/files/6238163/bc_mirror_memory.txt)

dillonhuff, 2021-03-31:
@jeffsetter would you mind adding these apps to `example_progs` instead of sending me the text files?

dillonhuff, 2021-04-01:
@jeffsetter so what are the concerns? Do the applications not run through the flow that you have used for other applications? If clockwork crashes on them could you send me the errors you see on your end?

jeffsetter, 2021-04-01:
My flow only runs to the CGRA, so I don't know if any run through the FPGA. The have the following errors:

bc_const:
```clockwork_codegen: prog.cpp:7811: isl_map* build_buffer_impl(prog&, UBuffer&, schedule_info&, ubuffer_impl&): Assertion `has_embarassing_partition' failed.```

bc_repeat:
```clockwork_codegen: prog.cpp:1305: std::map<std::__cxx11::basic_string<char>, UBuffer> build_buffers(prog&, umap*): Assertion `consumed_here != nullptr' failed.```

bc_mirror:
```clockwork_codegen: prog.cpp:1305: std::map<std::__cxx11::basic_string<char>, UBuffer> build_buffers(prog&, umap*): Assertion `consumed_here != nullptr' failed.```

When you run these three through the FPGA, what happens?