dillonhuff/clockwork #163: BFloat implementation

jeffsetter, 2021-03-31:
Several of my applications use bfloat types, including applications intended to test the bfloat hardware on the CGRA. What is the plan for supporting bfloat16_t?

dillonhuff, 2021-04-01:
@jeffsetter for the CGRA I have no idea. I assume the PE people are the ones to ask about that. From clockworks point of view a bfloat is just a 16 wide bit vector that gets moved between compute units, so it doesn't need any special treatment.

jeffsetter, 2021-04-01:
In order to run any pass through clockwork, I include the compute file. bfloat16_t is not recognized from the beginning, therefore, I can't run through any steps of clockwork.
```bin/fp_conv_2_2_compute.h:15:3: error: ‘bfloat16_t’ was not declared in this scope```