dillonhuff/clockwork #169: Gaussian unroll=4

jeffsetter, 2021-08-04:
I tried creating an unrolled version of gaussian that uses unroll=4. However, there seems to be an issue with the unoptimized code. Specifically with the error below. The app is here: https://github.com/dillonhuff/clockwork/blob/aha/example_progs/gaussian_unroll.cpp

```
cd ./bin && LD_LIBRARY_PATH=/nobackup/setter/h2h/hwbuffer/polyir/../clockwork/lib:/nobackup/setter/h2h/hwbuffer/polyir/../coreir/lib \
./clockwork_codegen unopt 1>mem_cout 2> >(tee -a mem_cout >&2); \
EXIT_CODE=$?; cd ..; \exit $EXIT_CODE
clockwork_codegen: ubuffer.cpp:358: std::__cxx11::string generate_multilinear_ram_addr(const string&, bank&, UBuffer&): Assertion `offsets.size() == 1' failed.
bash: line 2:  5642 Aborted                 (core dumped) LD_LIBRARY_PATH=/nobackup/setter/h2h/hwbuffer/polyir/../clockwork/lib:/nobackup/setter/h2h/hwbuffer/polyir/../coreir/lib ./clockwork_codegen unop$
../../hw_support/hardware_targets.mk:161: recipe for target 'bin/unoptimized_gaussian.cpp' failed
make: *** [bin/unoptimized_gaussian.cpp] Error 134
```

dillonhuff, 2021-08-06:
@jeffsetter the application seems to have piecewise access expressions, which is not currently supported:

https://github.com/dillonhuff/clockwork/blob/234955f3af53ff51d9f4acb30b91eeaa1b38b105/example_progs/gaussian_unroll.cpp#L128

jeffsetter, 2021-08-06:
Ah I see. I was able to resolve this in Halide by changing my unroll TailStrategy to use RoundUp.

closed 2021-08-06 by jeffsetter